garethky/PrusaSlicerPressureAdvanceCalibration #12: Support for OrcaSlicer?

kozross, 2023-12-22:
If I try to use this with Orca Slicer, I get issues with lots of missing settings. 
[image]

This information definitely exists in the gcode file, just under different names, and it would be really good to be able to use it.

garethky, 2023-12-22:
Orca slicer has these calibration patterns built in: https://github.com/SoftFever/OrcaSlicer/wiki/Calibration#Pressure-Advance

Why would you use this tool instead of that?

kozross, 2023-12-22:
Because I've never gotten them to work properly with my Mini.

garethky, 2023-12-22:
I wrote this for Prusa Slicer because I have an XL and an MK3/3.9.

Is there an issue open with Orca Slicer about Prusa Mini compatibility?

I'm not sure that working with all of the PS forks is something I want to take on. If forks change the gcode generator or don't take updates from upstream, this will turn into a lot of maintenance work.

Can you post the gcode to this issue so I can see how different it is?

kozross, 2023-12-22:
@garethky - here you go: [Prusa Slicer Pressure Advance Test_0.3mm_PLA_MINI_4m43s.zip](https://github.com/garethky/PrusaSlicerPressureAdvanceCalibration/files/13756679/Prusa.Slicer.Pressure.Advance.Test_0.3mm_PLA_MINI_4m43s.zip)