garethky/PrusaSlicerPressureAdvanceCalibration #13: Missing ;AFTER_LAYER_CHANGE comment.

mottld, 2023-12-28:
I get the following error when I upload the gcode to your site:

Could not find the last line of the start gcode block. Missing ;AFTER_LAYER_CHANGE comment. Check in the printers custom gcode settings.

What do I need to add? I am using Qidislicer which is just a rebranded prusaslicer.

Attached is my sliced gcode

[Prusa Slicer Pressure Advance Test.gcode.txt](https://github.com/garethky/PrusaSlicerPressureAdvanceCalibration/files/13790040/Prusa.Slicer.Pressure.Advance.Test.gcode.txt)

garethky, 2024-03-03:
In the "rebrand" they can make changes to all of these strings. Similarly #12 is because OrcaSlicer changed a number of the setting names in its output.

Keeping up with all of the forks is something I'm not sure I want to sign up for.

garethky, 2024-03-03:
Actually... I saw this error today... I dropped in the generated gcode from the Pressure Advance Tool back into the tool. Are you sure you are not doing that?

libluka, 2024-03-03:
@garethky I'm getting this error on the latest PrusaSlicer (2.7.2 Stable). Something seems to have changed where no ;AFTER_LAYER_CHANGE is written to the gcode. Instead, ;LAYER_CHANGE is inserted just before the gcode for the sliced test object.

```
M900 K0.07 ; Filament gcode
M572 S0.055 ; Filament gcode
M142 S36 ; set heatbreak target temp
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
M486 S0
G1 Z1.7 F720
G1 X124.328 Y185.828 F24000
G1 Z.2 F720
M73 P4 R4
G1 E1.5 F1500
M204 P500
;TYPE:Perimeter
;WIDTH:0.499999
```

I've attached the gcode of the sliced test object using the default 0.4 Prusa XL 5 Tool profile with Prusament PETG with the thumbnails removed for convenience. I tried the MK3 and it had the same behaviour, so I don't think the machine matters.
[Prusa Slicer Pressure Advance Test(2)_0.4n_0.2mm_PETG_XLIS_4m.gcode.txt](https://github.com/garethky/PrusaSlicerPressureAdvanceCalibration/files/14472219/Prusa.Slicer.Pressure.Advance.Test.2._0.4n_0.2mm_PETG_XLIS_4m.gcode.txt)

garethky, 2024-03-03:
Ahh, I got the new version and now I can reproduce it.

garethky, 2024-03-03:
@ libluka I've pushed an update that should fix this. Let me know if you have any issues now.

libluka, 2024-03-04:
@garethky seems to be fixed now! thanks for your work :)

closed 2024-03-04 by garethky