garethky/PrusaSlicerPressureAdvanceCalibration #15: ERROR "nil" is not a number

misisl, 2024-02-20:
Prusa Slicer 2.7.1,  Prusa XL 2T 
After importing gcode, your web show error warning and download button not work :(

[image]

[Pressure Advance Test_0.4n_0.2mm_PETG_XLIS_4m.gcode.txt](https://github.com/garethky/PrusaSlicerPressureAdvanceCalibration/files/14343494/Pressure.Advance.Test_0.4n_0.2mm_PETG_XLIS_4m.gcode.txt)

garethky, 2024-02-21:
Looks like some of these values are the string `nil` in the gcode file. I haven't seen that before! How do you get that to happen?

(I have seen Prusa Slicer use 0 to mean 'not set')

libluka, 2024-02-28:
I'm having the same issue slicing with a 5 Tool XL using all defaults with the input shaper printer profile.

libluka, 2024-03-03:
@garethky this seems to happen with multi-tool setups in PrusaSlicer where your extruders have different materials (eg. Prusa XL 5 head doing the test with Extruder 1 Siddament PETG but I have Prusament PETG on E2, Prusament PLA on E3, Siddament PETG on E4 and Generic Flex on E5).

Setting all the extruders in PrusaSlicer to be Prusament PETG for example completely removes parameters such as filament_retract_speed from being written to the gcode, therefore no 'nil' values.

Also as a side note, issue #13 - maybe it's been a PrusaSlicer update or something, but as far as I can see the beginning of the object is marked by ;LAYER_CHANGE rather than ;AFTER_LAYER_CHANGE. Since the test file is a single layer, there is never an ;AFTER_LAYER_CHANGE comment added.

garethky, 2024-03-03:
@misisl can you please test and let me know if things work as you expected. I was able to solve the issue but only able to test with the file you provided (plus unit testing).

closed 2024-03-04 by garethky