garethky/PrusaSlicerPressureAdvanceCalibration #16: Invalid value for zhop instruction

mayty, 2024-02-28:
Zhop instruction needs to be `G1 Z{layer_height + z_hop}`.
When I ran the generated code with layer height 0.4mm and zhop 0.25mm nozzle ran through printed layers (see attached image)
[image]

garethky, 2024-03-03:
Yes, the z-hop height isn't being added to the layer height.

garethky, 2024-03-03:
There were cases where it never went back down to the regular z height. There were also many places where there were long travel moves happened with no z hop.

@mayty please try out the new version and see if this fixes your issue.

mayty, 2024-03-08:
@garethky 
It works like a charm, thanks

closed 2024-07-05 by garethky