garethky/PrusaSlicerPressureAdvanceCalibration #17: Calibration with Filaments which need slow speeds/are limited by volumetric flow rather than print speed

Unostot, 2024-04-09:
Hello,

thanks for this nice tool...it worked good for "regular" filament...

Now I wanted to try it with flexible filament, but ran into a slight problem between your tool and the way prusa slicer works...
A few years ago I used cura and only a single hotend...the world was pretty simple and I configured all profiles as needed...

then I switched for the XL to prusa slicer, and was a bit confused:
There seems no way, to set print speeds per filament, exept fiddling with the max volumetric flow...
this might work nice as soon as everything is correctly configured, but until then it gives problems with multiple extruders:
* print speeds can only be set for the full print
* speed per filament needs to be controlled via volumetric limits in the filament settings

your tool reads the the speed from the gcode, but gets some things partially wrong (ok, technically it reads them right, but the result is producing a at best inverted test, in the worst case jammed extruders due to too fast  slow print speeds :) )
[image]

As you can see, it reads the speeds from the print profile, which are quite high...
it also sets fast speed to the max flow configured for the filament (the 14.443 mm/s, which if I'm not mistaken comes from the 1.3 mm3/s configured for the filament)

now this leads to a print which uses the printspeed slow 40mm/s and for fast the limited speed (in this case 14.443)...and my extruder gets jammend because its way too fast for the filament which is loaded...

I'm not sure how it could be improved. I think I understand how you work (not yet looked at the code though)....and this way works for filaments which can be printed with those speeds and usually not limited via volumetric flow...but for filaments which have a volumetric limit resulting in slower speeds then the normal printspeed it leads to "inverted speeds" 

I also know I could work around it, by specifying speeds for this print different (I think it would be enough to set the first layer slow enough)

But it would be nice to find a way, which extracts the actual limit speeds from the gcode from the file, especially for users which are e.g. new to calibrating stuff...especially when you write "slice it with your normal printsettings, the tool takes care of the rest"

Since you already take the maximum speed from the volumetric speed, perhaps a option/logic would be a way, so if "fast speed > normal printspeed" -> "set normal printspeed to <insert random wisely chosen lower number here>"

Hope we can improve this, since pressure advance is especially nice to set correct on the complicated filaments,
so those cases should work, too...

Greetings,
Uno

P.S. another minor issue: when slicing it for extruder 5 but have different filament set for extruder 1, the tool extracts at least the name from extruder 1...not checked if other options are too read wrong, like perhaps the volumetric limit)

garethky, 2024-04-09:
Right, only the high speed has the volumetric flow limit applied. If the limit was applied to all extrusion speeds, then all of the speeds would be the same in your example. I.e. there would be no speed change in the test, making the test useless.

I could see adding a safety check to see if the max speed is lower than the low speed and then refusing to generate a test file in that case. Ideally saving you from a jammed nozzle. Safer, but that doesn't get you a working test. 

There are other versions of this test that don't rely on their being 2 speeds, they look at the acceleration around a corner in the geometry. This might be a good use case for that type of test.

garethky, 2026-02-15:
### TODO:
1) Check that the regular print speed is not being limited by the volumetric flow rate, if it is show an error
2) If the slow speed and fast speed in the test end up being the same, show the user an error.

closed 2026-02-17 by garethky