garethky/PrusaSlicerPressureAdvanceCalibration #20: [QUESTION] M900 vs. M572 - Prusa MK3.5 Firmware 6.0.3

artistassembly, 2024-07-05:
Hello,

Thank you for creating & sharing this!  I generated a calibration gcode file for the Prusa MK3.5 and it appears to be incrementing the M900 K values.  The default MK3.5 printer & PETG filament profiles in Prusa Slicer 2.8.0 do not use M900 K values.  The default profiles are setting M572 S values instead.  This makes me think the calibration gcode cannot be used to calibrate the Pressure/Linear Advance for the MK3.5.  Please correct me if my understanding is inaccurate.

The calibration generator at https://garethky.github.io/PrusaSlicerPressureAdvanceCalibration/ shows:
```
Filament
Printer | MK3.5 | printer_model

Pressure Advance
Pressure Advance Type | Marlin Linear Advance: M900 | Selected based on gcode_flavor:marlin2 and printer_model:MK3.5
```
It looks like the calibration generator is recognizing the printer as a MK3.5 but may not be using the correct gcode_flavor, or Prusa's variant of it.

Is there a way to generate the test gcode to calibrate pressure advance using M572 instead of M900?

garethky, 2024-07-05:
This may have changed recently. M900 is the original GCode for pressure advance on the 8bit boards and M572 is the gcode on the new 32bit. You can see how this used to work in this thread: https://forum.prusa3d.com/forum/general-software-discussion/m900-vs-m572/
Every printer got M900 except `XLIS` and `MK4IS`.

In Prusa Slicer 2.8 I see this:

```
{if printer_notes!~/.*(MK3.5|MINIIS).*/}
M900 K{if printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.6}0.12{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.8}0.06{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/}0.2{elsif nozzle_diameter[0]==0.8}0.02{elsif nozzle_diameter[0]==0.6}0.04{else}0.08{endif} ; Filament gcode LA 1.5
{if printer_notes=~/.*PRINTER_MODEL_MINI.*/};{elsif printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}M900 K200{elsif nozzle_diameter[0]==0.6}M900 K24{elsif nozzle_diameter[0]==0.8};{else}M900 K45{endif} ; Filament gcode LA 1.0
{endif}

{if printer_notes=~/.*MINIIS.*/}
M572 S{if nozzle_diameter[0]==0.6}0.22{elsif nozzle_diameter[0]==0.8}0.15{elsif nozzle_diameter[0]==0.4}0.36{elsif nozzle_diameter[0]==0.25}1.02{else}0{endif}
{endif}

{if printer_notes=~/.*MK3.5.*/}
M572 S{if nozzle_diameter[0]==0.4}0.052{elsif nozzle_diameter[0]==0.5}0.042{elsif nozzle_diameter[0]==0.6}0.025{elsif nozzle_diameter[0]==0.8}0.018{elsif nozzle_diameter[0]==0.25}0.18{elsif nozzle_diameter[0]==0.3}0.1{else}0{endif} ; Filament gcode
{endif}
```

artistassembly, 2024-07-05:
Thank you for taking the time to reply @garethky!  The MK3.5 is definitely the new 32 bit board, not the 8bit board.  I believe the MK3.5 firmware is more closely related to the MK4 than the MK3S+.

I see the same filament start g-code in Prusa Slicer 2.8.0.  I interpret them as follows:

This if statement is true if the printer notes **DO NOT** contain MK3.5 or MINIIS.
```
{if printer_notes!~/.*(MK3.5|MINIIS).*/}
M900 K{if printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.6}0.12{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.8}0.06{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/}0.2{elsif nozzle_diameter[0]==0.8}0.02{elsif nozzle_diameter[0]==0.6}0.04{else}0.08{endif} ; Filament gcode LA 1.5
{if printer_notes=~/.*PRINTER_MODEL_MINI.*/};{elsif printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}M900 K200{elsif nozzle_diameter[0]==0.6}M900 K24{elsif nozzle_diameter[0]==0.8};{else}M900 K45{endif} ; Filament gcode LA 1.0
{endif}
```

This statement is true if the printer notes contain MINIIS
```
{if printer_notes=~/.*MINIIS.*/}
M572 S{if nozzle_diameter[0]==0.6}0.22{elsif nozzle_diameter[0]==0.8}0.15{elsif nozzle_diameter[0]==0.4}0.36{elsif nozzle_diameter[0]==0.25}1.02{else}0{endif}
{endif}
```

This statement is true if the printer notes contain MK3.5
```
{if printer_notes=~/.*MK3.5.*/}
M572 S{if nozzle_diameter[0]==0.4}0.052{elsif nozzle_diameter[0]==0.5}0.042{elsif nozzle_diameter[0]==0.6}0.025{elsif nozzle_diameter[0]==0.8}0.018{elsif nozzle_diameter[0]==0.25}0.18{elsif nozzle_diameter[0]==0.3}0.1{else}0{endif} ; Filament gcode
{endif}
```

If my interpretation is correct, the 3rd if statement would be the only one resolving as true for the MK3.5.  This uses the M572 S values & corresponds with what I see in the gcode generated by Prusa Slicer.

Let me know if I can provide any additional info.  I'm happy to run commands against my MK3.5 if that would be helpful.

garethky, 2024-07-06:
The MK3.5 is, for sure, the 8 bit board. I cant find where its firmware handles gcode: https://github.com/prusa3d/Prusa-Firmware

In the 32dbit firmware the M900 command has been re-wired to just call M572:
https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/ce6d58b80c995dd8d5923d39688b195a6ac10b1e/lib/Marlin/Marlin/src/gcode/feature/advance/M900.cpp#L133

artistassembly, 2024-07-06:
I must respectively disagree with you about which board is in the MK3.5.  I recently upgraded my MK3S+ to the MK3.5 & the board was one of the components replaced.  The MK3.5 has the full color display with touch controls.

In the 1st link you posted, note the list of printers that firmware supports.  It does not include the MK3.5 nor MK3.9.

> This repository contains the source code and the development versions of the firmware running on the [Original Prusa i3](https://prusa3d.com/) **MK3S/MK3/MK2.5S/MK2.5** line of printers.

The xBuddy firmware is what is on the MK3.5  https://github.com/prusa3d/Prusa-Firmware-Buddy.  While the readme on that page does not included the MK3.5, I believe that is simply an oversight.  

The MK3.5 is mentioned in the latest release 6.0.3 https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.3.  The MK3.5 is also listed as using the Buddy firmware on the Prusa download page https://help.prusa3d.com/downloads.

The Buddy firmware 6.0.3 is used on the MK4, MK3.9, MK3.5, MINI & XL per the firmware release notes.

Below is a photo of the system info screen from my Mk3.5.
[image]

Let me know if there is additional info I may provided.

artistassembly, 2024-07-06:
I would also note the credits of the buddy firmware readme.md https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/README.md

```
Credits

    [Marlin](https://marlinfw.org/) - 3D printing core driver
    [Klipper](https://www.klipper3d.org/) - input shaper code based on Klipper
```

This indicates to me that the bulk of the buddy firmware is marlin, but the input shaper code is based upon Klipper which uses M572.

garethky, 2024-10-11:
Sorry this took so long to fix. Live has been really busy. I also wanted to really think this through and write tests for it so the next printer/upgrade they launch doesn't get the wrong value.

garethky, 2024-10-11:
Please give the fix a go

artistassembly, 2024-10-30:
Thanks so much for the update!  I have also been busy with other things but hope to get back to the printer soon.  I'll give this a test ASAP & let you know how it goes.

closed 2026-02-15 by garethky

artistassembly, 2026-09-20:
I know it's been a long time, but I finally got back to this & it worked as expected on the MK3.5.  Thanks again!