garethky/PrusaSlicerPressureAdvanceCalibration #23: Receiving error "Could not find the first line of the end gcode block. Missing ; Filament-specific end gcode comment."; however, the line is in the g-code

SpoolItUp, 2024-12-13:
As mentioned in the title, I am getting the error that the "; Filalament-specific end gcode comment" is missing; however, it's in the the g-code. Sliced on Prusaslicer 2.8.1.

garethky, 2024-12-23:
Looks like a space has been added at the end of the line:

Original:
`; Filament-specific end gcode\n;`

New: 
```
; Filament-specific end gcode \n;
                             ^
```

The lines are not exactly the same so they fail to match. I could make this check more lenient by allowing additional whitespace at the end of the line.

Before deciding what to do, I would really like to understand is how you got this extra space in there? I have 2.9.0 and I don't see it in my sliced GCode files with the Prusament profiles. What filament profile is this?

krom, 2024-12-23:
> Looks like a space has been added at the end of the line:

Thank you, this helped

> how you got this extra space in there?

I Imported custom profile provided by my teammate at least one year ago

also I added manually to .gcode fille
```
printer_model = Kingroon KP3S Pro V2
```
Because my Slicer don't provided this value
And I got error
>Test pattern generation failed: No max value found, all values are null or 0!

And I guess it's because this printer is not supported but it uses standard klipper commands