garethky/PrusaSlicerPressureAdvanceCalibration #24: Numbers are Messy

garethky, 2025-01-26:
Printing individual numbers is messy. Removing them from the build sheet causes them to spray everywhere. Depending on the filament they can be difficult to remove from the build sheet. For certain filament it might be a good idea to cancel the print before the numbers get printed. 

Ideas:
- Show a graphical representation of the printed lines in the web tool so the user can identify the PA value for a line without the printed labels.
- Add a backing layer below the numbers so they are easy to remove as a single piece. (Orca Slicer does this)
- Add a pause between printing the test pattern and printing the numbers so the user can cancel the print to resume it if they want the numbers. A message (M117?) can tell the user "resume to print the legend".
- give the user 3 options for generating the gcode: print the legend, pause before printing legend, don't print the legend.

Pausing has downsides, especially if the user isn't expecting it to happen and they wander off. I think that has to be an opt-in experience from a safety perspective. But I think it would be OK to save that choice in a cookie for the future.

JulianSinulf, 2025-03-21:
Hi!
I vote for the backing layer.
I've already printed from Orca.
I've done it without numbers, looking at the web interface, or rather in the slicer (orca), but the many lines can be quite distracting.
An extra layer under the numbers could be good. With the addition that maybe the value would be there for every line, but in this case the numbers could be shifted. For example, for the first line, there would be 0.00 in the first column, for the second line, the number would have a second column and the number would be shifted a little higher, while the number of the third line would be back in the first column in the right place, as before. If I remember correctly, there is only a marking for odd lines.
I don't want to do it from Orca because the printer parameters don't match the parameters in PrusaSlicer.

garethky, 2026-02-15:
I came to the conclusion that if the experience without the numbers was easy to use I would personally prefer that because its fast to print. I was just canceling the print when the numbers started to avoid them.

So this change will:

* Add a check box to disable printing of the numbers. This is remembered across sessions as a preference.
* Add tick marks next to each line so they are easier to count, just like how a ruler is marked
* Add a UI diagram that shows the line numbers, PA values and the tick marks for easy reference
* Let you click the PA value to copy the gcode for that value right to the clipboard without filling out the form at the bottom of the page

[image]

[image]